bluez/bluez #1500: Rpi 5 not discovering the ble devices,discovering only classic devices

harish-slx, 2025-08-12:
During extended tests with the external adapter, after a day or two it enters a bad state and detects only BR/EDR (Classic) devices, failing to discover BLE devices. Testing was performed with a TP-Link UB500 USB Bluetooth adapter. Because the Raspberry Pi 5’s built-in Bluetooth does not support LE extended scanning, I must use the external adapter. I need a permanent fix that does not require restarting Bluetooth, resetting the adapter, or power-cycling the device.

BlueZ version - 5.66
OS - Bookworm 
kernel version - 6.12.25+rpt-rpi-2712

Vudentz, 2025-08-12:
@harish-slx I would start by updating the BlueZ version, not idea why some distro would ship an older version of BlueZ with a newer kernel. Also, it would probably help if the btmon traces are attached to see what is going at the HCI level.

harish-slx, 2025-08-19:
@ammarkh95 Did you run test for a day or two?
The device is discovered for certain period, but I'm asking about if the test run for a day or two, rpi5 detects only BR/EDR devices, failing to discover the BLE devices.

closed 2025-10-01 by github-actions[bot]