bluez/bluez #1545: A2DP devices disconnect immediately after connecting due to race condition between PipeWire/WirePlumber and BlueZ

CaCTuCaTu4ECKuu, 2025-09-05:
- PipeWire version: 1.4.7
- Distribution and distribution version: Fedora Linux 42 (KDE Plasma Desktop Edition)
- Desktop Environment: KDE Plasma
- Kernel version: 6.16.3-200.fc42.x86_6
- BlueZ version: bluetoothctl: 5.83
- `lsusb`:

```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 05e3:0761 Genesys Logic, Inc. Genesys Mass Storage Device
Bus 003 Device 003: ID 3277:007e Sonix Technology Co., Ltd. Usb Camera
Bus 003 Device 004: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

- Bluetooth devices:

```
Device 41:42:53:7A:A2:17 ● Commo One
Device C5:47:CA:55:D6:B9 G603
```

## Description of Problem:

When connecting Bluetooth headphones, the device connects but disconnects immediately.\
From the logs, it seems that PipeWire/WirePlumber tries to release the `MediaTransport` for A2DP, but BlueZ returns an error: the `Release` method is not implemented for this device/BlueZ version.

This creates a race condition:

* The device connects,
* PipeWire/WirePlumber attempts to set up the transport,
* BlueZ rejects the `Release` call,
* The device falls back to HSP/HFP profile, then disconnects,
* Endpoints are unregistered and registered again.

After manually restarting the `bluetooth` service (`systemctl restart bluetooth`), everything works fine and the A2DP connection remains stable.

This strongly suggests a startup ordering issue between BlueZ and PipeWire/WirePlumber.

## How Reproducible:

### Steps to Reproduce (Presumably):

1. Fresh install Fedora 42 KDE.
2. Update everything (through Discover).
3. Add a Bluetooth headset device (with A2DP profile support).
4. Pair headset device.
5. Attempt to connect to the device.
6. Observe that the device connects briefly and then disconnects right away.

### Actual Results:

Devices disconnect immediately unless the bluetooth service is restarted.

### Expected Results:

Bluetooth audio devices should connect reliably to A2DP after system startup, without requiring a manual restart of the bluetooth service.

# Additional Info (as attachments):

- Hardware: TECNO MegaBook T1 (Intel Wireless-AC 9461D2W), Commo One headphones
- `bluetooth.service`: [bluetooth.service.log](https://github.com/user-attachments/files/22173038/bluetooth.service.log)
- aplay -l: [aplay_output.txt](https://github.com/user-attachments/files/22173041/aplay_output.txt)
- `wireplumber.service`: [wireplumber.service.log](https://github.com/user-attachments/files/22173043/wireplumber.service.log)

Issue from PipeWire tracker: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4882

borine, 2025-09-07:
The symptoms described here look very much like issue #1471. That one is not using pipewire, but has the same problem that the A2DP profile is dropped leaving only HFP, and the problem does not exist in bluez 5.82. I don't know if the underlying cause is the same (not knowing too much about how pipewire configures its A2DP endpoints); but a btmon log of the connection negotiation would help to diagnose it.

pv, 2025-09-07:
btmon traces are needed here

CaCTuCaTu4ECKuu, 2025-09-07:
I can reproduce the problem, can you guide me what exactly I have to do to get right data logs?

PranavBhattarai, 2025-09-18:
@CaCTuCaTu4ECKuu I think using `sudo btmon` command in the terminal, do the the steps what you did to generate the Bluetooth logs. Steps like turning on the Bluetooth, turning off the Bluetooth, **playing the music and waiting for A2DP to fail to send the packets**, pushing the music, switching Bluetooth profile, etc.

CaCTuCaTu4ECKuu, 2025-09-19:
My file is pretty big, I guess it's becouse this time I play audio in bg.
I restarted the PC, prepared everyting, started btmon `sudo btmon > /home/user/btmon.log` and clicked **connect** to device on widget.
Headphones connected, if you have audio playing then headphones pick it up and play, but in the end after few seconds it disconnects.

[btmon.log](https://github.com/user-attachments/files/22426102/btmon.log)

anakinrpi, 2025-09-29:
I’m leaving here, in case it’s useful, a bug report in Fedora’s Bugzilla where the relationship between KDE and BlueZ regarding this problem is analyzed: https://bugzilla.redhat.com/show_bug.cgi?id=2397966

Vudentz, 2025-10-01:
```
> HCI Event: Link Key Request (0x17) plen 6               #546 [hci0] 12.071203
        Address: 41:42:53:7A:A2:17 (OUI 41-42-53)
< HCI Command: Link Key Request N.. (0x01|0x000c) plen 6  #547 [hci0] 12.071227
        Address: 41:42:53:7A:A2:17 (OUI 41-42-53)
> HCI Event: Command Complete (0x0e) plen 10              #548 [hci0] 12.072212
      Link Key Request Negative Reply (0x01|0x000c) ncmd 1
        Status: Success (0x00)
        Address: 41:42:53:7A:A2:17 (OUI 41-42-53)
...
> HCI Event: Simple Pairing Complete (0x36) plen 7        #661 [hci0] 13.189271
        Status: Authentication Failure (0x05)
        Address: 41:42:53:7A:A2:17 (OUI 41-42-53)
> HCI Event: Auth Complete (0x06) plen 3                  #662 [hci0] 13.190256
        Status: Authentication Failure (0x05)
        Handle: 256 Address: 41:42:53:7A:A2:17 (OUI 41-42-53)
@ MGMT Event: Authentication Failed (0x0011) plen 8   {0x0001} [hci0] 13.190285
        BR/EDR Address: 41:42:53:7A:A2:17 (OUI 41-42-53)
        Status: Authentication Failed (0x05)
```

Is the device even able to pair at all? At least from the above it doesn't look so, the weird thing is that this appears out of nowhere when A2DP is already streaming.

CaCTuCaTu4ECKuu, 2025-10-01:
Device not only connects, but also output audio. I run video in background and press "connect", headphones connected and I was hearing audio but only for a few seconds before it disconnects
From what I see in other threads it have something to do with KDE

Hi-Angel, 2025-11-08:
@anakinrpi I created [this issue](https://github.com/bluez/bluez/issues/1646), for people seeing exactly the same behavior, but no "Authentication Requested" in the dbus log.

@CaCTuCaTu4ECKuu I see you didn't provide the dbus logs, presumably because the issue disappeared? If so, I think you should be able to reproduce it again by removing the headphones from the known devices.

fdanis-oss, 2025-12-15:
Afaict from the logs, I agree with comment https://github.com/bluez/bluez/issues/1545#issuecomment-3601654635 stating that the second authentication  request seems strange.

If you're able to reproduce it can you enable BlueZ daemon logs by adding `-d` to the `bluetoothd` command then run the following commands:
- `sudo busctl monitor org.bluez  > dbus.log`
- `sudo btmon -w btsnoop.log`

Then try to pair with your device.

Then please copy the `bluetoothd` logs, `dbus.log` and `btsnoop.log`