bluez/bluez #1570: A2DP connection fails on startup with BlueZ 5.83 and 5.84 (Regression)

anakinrpi, 2025-09-24:
**Summary:**
On Fedora 42, **BlueZ versions 5.83 and 5.84** fail to connect my **Redmi Buds 6 Lite** and **Redmi Buds 5** to the A2DP profile on system startup. The issue affects multiple codecs (SBC, AAC, SBC-XQ) and causes audio instability. This is a regression, as the problem is resolved by downgrading to **BlueZ 5.80**, which is the only older version available for testing in the Fedora repositories.

**Description:**
When I boot my PC with my previously paired **Redmi Buds 6 Lite** or **Redmi Buds 5** powered on, the device fails to connect to the A2DP profile. The system attempts to connect via the HSP profile instead, leading to the errors seen in `journalctl`. The issue is not isolated to one specific codec; it affects A2DP functionality regardless of whether the preferred codec is SBC, AAC, or SBC-XQ. This results in the following problems:
1.  **Connection Failure on Startup:** The headphones do not connect automatically.
2.  **Audio Instability:** Even after manual connection, the audio is unstable, with one earbud occasionally losing sound before stabilizing.
3.  **Required Manual Intervention:** I have to manually reconnect the device via KDE's Bluetooth settings or restart the Bluetooth service to get a stable connection.

These issues do not occur when using BlueZ 5.80.

**System Information:**
* **Distribution:** Fedora 42 KDE
* **BlueZ version (with the error):** 5.84 (and verified with 5.83)
* **BlueZ version (working correctly):** 5.80
* **Bluetooth Adapter:** TP-Link UB500 Plus (Chipset: Realtek RTL8761BU)
* **Bluetooth Devices:** Xiaomi Redmi Buds 6 Lite and Redmi Buds 5
* **Affected Codecs:** SBC, AAC, SBC-XQ

**Steps to Reproduce:**
1.  Have a PC running Fedora with a BlueZ version of 5.83 or newer.
2.  Pair either the **Redmi Buds 6 Lite** or **Redmi Buds 5**.
3.  Shut down the PC and ensure the earbuds are turned on.
4.  Boot the PC. The earbuds will fail to connect automatically to the A2DP profile.

Actual Results:
The earbuds do not connect automatically. The journalctl logs show the following errors:
sep 24 16:28:45 Ryzen-PC bluetoothd[1188]: Failed to set mode: Not Supported (0x0c)
sep 24 16:29:05 Ryzen-PC bluetoothd[1188]: src/device.c:search_cb() 68:4D:B6:08:80:E3: error updating services: Input/output error (5)
sep 24 16:31:09 Ryzen-PC bluetoothd[1188]: No matching connection for device
sep 24 16:31:24 Ryzen-PC bluetoothd[1188]: src/device.c:search_cb() 68:4D:B6:08:80:E3: error updating services: Input/output error (5)
sep 24 16:31:24 Ryzen-PC bluetoothd[1188]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
sep 24 16:56:04 Ryzen-PC systemd[1]: Stopping bluetooth.service - Bluetooth service...
sep 24 16:56:04 Ryzen-PC bluetoothd[1188]: Terminating
sep 24 16:56:04 Ryzen-PC bluetoothd[1188]: src/bearer.c:btd_bearer_disconnected() Unknown disconnection value: 21
sep 24 16:56:04 Ryzen-PC bluetoothd[1188]: src/device.c:device_disconnected() Unknown disconnection value: 21
sep 24 16:56:04 Ryzen-PC bluetoothd[1188]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
sep 24 16:56:39 Ryzen-PC bluetoothd[1193]: Failed to set mode: Not Supported (0x0c)
sep 24 16:56:58 Ryzen-PC bluetoothd[1193]: src/device.c:search_cb() 68:4D:B6:08:80:E3: error updating services: Input/output error (5)

**Logs from `journalctl`:**
sep 24 16:28:45 Ryzen-PC systemd[1]: Started bluetooth.service - Bluetooth service.
sep 24 16:28:45 Ryzen-PC bluetoothd[1188]: Starting SDP server
sep 24 16:28:45 Ryzen-PC bluetoothd[1188]: Bluetooth management interface 1.23 initialized
sep 24 16:28:45 Ryzen-PC bluetoothd[1188]: Battery Provider Manager created
sep 24 16:28:45 Ryzen-PC bluetoothd[1188]: Failed to set mode: Not Supported (0x0c)
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/ldac
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSink/aptx_hd
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/aptx_hd
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSink/aptx
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/aptx
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSink/aac
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/aac
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSink/opus_g
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/opus_g
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSink/sbc
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/sbc
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/aptx_ll_1
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/aptx_ll_0
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/faststream
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/faststream_duplex
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSink/opus_05
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/opus_05
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSink/opus_05_duplex
sep 24 16:28:53 Ryzen-PC bluetoothd[1188]: Endpoint registered: sender=:1.50 path=/MediaEndpoint/A2DPSource/opus_05_duplex
sep 24 16:29:05 Ryzen-PC bluetoothd[1188]: src/device.c:search_cb() 68:4D:B6:08:80:E3: error updating services: Input/output error (5)
sep 24 16:31:09 Ryzen-PC bluetoothd[1188]: No matching connection for device
sep 24 16:31:24 Ryzen-PC bluetoothd[1188]: src/device.c:search_cb() 68:4D:B6:08:80:E3: error updating services: Input/output error (5)
[... 37 lines cut ...]
sep 24 16:56:39 Ryzen-PC bluetoothd[1193]: Bluetooth management interface 1.23 initialized
sep 24 16:56:39 Ryzen-PC bluetoothd[1193]: Battery Provider Manager created
sep 24 16:56:39 Ryzen-PC bluetoothd[1193]: Failed to set mode: Not Supported (0x0c)
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/ldac
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSink/aptx_hd
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/aptx_hd
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSink/aptx
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/aptx
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSink/aac
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/aac
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSink/opus_g
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/opus_g
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSink/sbc
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/sbc
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/aptx_ll_1
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/aptx_ll_0
[... cut ...]
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSink/opus_05
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/opus_05
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSink/opus_05_duplex
sep 24 16:56:47 Ryzen-PC bluetoothd[1193]: Endpoint registered: sender=:1.51 path=/MediaEndpoint/A2DPSource/opus_05_duplex
sep 24 16:56:58 Ryzen-PC bluetoothd[1193]: src/device.c:search_cb() 68:4D:B6:08:80:E3: error updating services: Input/output error (5)

anakinrpi, 2025-09-29:
I’m sharing the progress found in Fedora’s Bugzilla; it seems to be related to versions 5.83+ only in the KDE environment:

-------------
anakinrpi 2025-09-27 17:29:00 UTC
I think this discussion forum thread is related to my error: https://discussion.fedoraproject.org/t/bluetooth-service-needs-to-be-restarted-to-work/156298
-------------

Comment 14Fedora Update System 2025-09-28 01:03:35 UTC
FEDORA-2025-1c4ded4bf6 (bluez-5.84-2.fc43) has been pushed to the Fedora 43 stable repository.
If problem still persists, please make note of it in this bug report.
-------------

Comment 15Peter Robinson 2025-09-28 09:52:45 UTC
> It’s worth noting that version 5.80 works correctly for me.

Please test the intervening builds to determine what the last working/first failed builds were, they can be retrieved from koji. To date you're the only person who's reported issues (and audio issues are reported basically the moment they regress) so there's something quite specific in your setup.

https://koji.fedoraproject.org/koji/packageinfo?packageID=6777
-------------

Comment 16anakinrpi 2025-09-28 13:41:47 UTC
Hello, thanks for the response!
Version 5.80-1.fc42 works fine
Version 5.81-2.fc42 works fine
Version 5.83-1.fc42 errors start when connecting the device at system startup
Version 5.83-2.fc42 same problem
Version 5.84-1.fc42 same problem
Version 5.84-2.fc42 same problem

Regarding the audio quality issues, I’m testing whether it has to do with having been using the SBC-XQ codec, which the headphones might not fully support. I’m trying AAC, and so far I haven’t had any problems.
-------------

Comment 17Peter Robinson 2025-09-28 14:26:23 UTC
> Regarding the audio quality issues, I’m testing whether it has to do with
> having been using the SBC-XQ codec, which the headphones might not fully
> support. I’m trying AAC, and so far I haven’t had any problems.

That would be a different bug report with pipewire, as bluez doesn't deal directly with the audio streams, but reading up SBC-XQ looks to be somewhat of a hack [1] anyway so you may be better off with AAC.

[1] https://soundexpert.org/articles/-/blogs/audio-quality-of-sbc-xq-bluetooth-audio-codec
-------------

Comment 18Peter Robinson 2025-09-28 14:35:04 UTC
(In reply to anakinrpi from comment #13)
> I think this discussion forum thread is related to my error:
> https://discussion.fedoraproject.org/t/bluetooth-service-needs-to-be-
> restarted-to-work/156298

And reading through that thread shows it seems to only be happening on KDE and not other desktop environments, which likely explains why I've not seen it reported until now, now the question is what KDE does differently to the A2DP that GNOME and other desktops don't that makes bluez misbehave.
-------------

Comment 19Peter Robinson 2025-09-29 09:22:31 UTC
So yesterday digging through a few posts on the forum from June when 5.83 landed there's at least a few that mention issues when used with KDE:

https://discussion.fedoraproject.org/t/bluetooth-service-needs-to-be-restarted-to-work/156298
https://discussion.fedoraproject.org/t/bluetooth-is-extremely-bugged-out-in-f42-kde-also-in-f41-kde/158644

So it looks like some change in bluez that affects KDE in particular but isn't necessarily seen by gnome or other desktop users.

anakinrpi, 2025-09-29:
Here is a similar issue reported in CachyOS:"
https://discuss.cachyos.org/t/issue-with-bluetooth-headphone-disconnecting/14952

CaCTuCaTu4ECKuu, 2025-09-30:
I think we have the same issue here -> https://github.com/bluez/bluez/issues/1545#issuecomment-3348331737

hadess, 2025-10-08:
You might want to test with:
https://github.com/bluez/bluetooth-next/commit/014b7133fa7b534d9149ae5126bbb9f4434e4ebb

anakinrpi, 2025-10-22:
I just installed 6.17.4-200.fc42.x86_64, and the problem persists; however, with bluez 5.81-2, it works perfectly.

anakinrpi, 2025-10-22:
I have pipewire-1.4.9-1.fc42.x86_64 and pipewire-1.4.9-1.fc42.i686 installed, and unfortunately the problem still persists. Thanks for your input.

oznogon, 2025-12-31:
This also affects me in Fedora 43 KDE with Shokz OpenMove and Sony WH-XB910N headsets Logs are similar to anakinrpi's. Restarting the bluetooth service (`sudo systemctl restart bluetooth`) temporarily fixes the issue.

I exclusively used AAC prior to this issue on both headsets. Installed Pipewire is 1.4.9-1.fc43.

fskoras, 2026-01-11:
I have similar issue with Soundcore Q30 headphones. `sudo systemctl restart bluetooth` after startup solves the problem but it has to be done each restart. I love fedora KDE 43 but this driver issue is pretty annoying :)

YuvamGupta, 2026-02-15:
Hi everyone
I am an arch user (hyprland) and I still had this issue. I have some components of KDE from kde connect and dolphin. In my case, `sudo systemctl restart bluetooth` didn't work. I tried everything from reinstalling arch to using the `linux-lts` kernel.
If you can't get a permanent fix by restarting bluetooth, the best way is to just **downgrade bluez and bluez utils**. 
Arch guide
- `sudo pacman -S downgrade` chaotic-aur repo required
- `sudo downgrade bluez bluez-utils` choose the latest version that works for you
- Add bluez and bluez utils to Ignore packages in `pacman.conf`

I hope that this issue is fixed by KDE/bluez since a large share of linux desktops require KDE or its derivatives.

**Keep checking the latest version periodically to see if they have fixed it. If their is a fix, REMOVE FROM IGNORE PACKAGE**

yaomtc, 2026-09-07:
I am on Fedora with a Thinkpad T41 Gen2 with a Realtek Bluetooth chip, and I have this issue with my Soundcore Life Q30 headphones... and with bluez 5.80!

yaomtc, 2026-09-09:
I thought the removal of the stale label was an automatic thing? github-actions applied the label, removed it, and added it again, but it seems unable to remove it anymore

Journal output from pairing, and then manually reconnecting a few minutes later:

```
Sep 07 16:15:51 fedora systemd[2027]: Reached target bluetooth.target - Bluetooth.
Sep 07 16:15:52 fedora bluetoothd[1437]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 98:47:44:A3:1A:DE: Connection refused (111)
Sep 07 16:15:52 fedora bluetoothd[1437]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 98:47:44:A3:1A:DE: Connection refused (111)
Sep 07 16:15:52 fedora plasmashell[2555]: PendingCall Error: "br-connection-refused"

Sep 07 16:16:03 fedora plasmashell[2555]: qrc:/qt/qml/org/kde/bluedevil/components/ForgetDeviceDialog.qml:38: TypeError: Cannot read property 'name' of null
Sep 07 16:16:03 fedora plasmashell[2555]: qrc:/qt/qml/org/kde/bluedevil/components/ForgetDeviceDialog.qml:89: TypeError: Cannot read property 'adapter' of null

Sep 07 16:16:14 fedora systemd[2027]: Started app-org.kde.bluedevilwizard@038ec0bfa8394f4d9707d2661115c76b.service - Add Bluetooth Device - Add Bluetooth Device.

Sep 07 16:16:23 fedora kernel: input: soundcore Life Q30 (AVRCP) as /devices/virtual/input/input25
Sep 07 16:16:23 fedora systemd-logind[1477]: Watching system buttons on /dev/input/event19 (soundcore Life Q30 (AVRCP))
Sep 07 16:16:23 fedora bluedevil-wizard[4904]: Had queued notifications on destruction. Was the eventloop running?
Sep 07 16:16:23 fedora systemd[2027]: app-org.kde.bluedevilwizard@038ec0bfa8394f4d9707d2661115c76b.service: Main process exited, code=exited, status=1/FAILURE
Sep 07 16:16:23 fedora systemd[2027]: app-org.kde.bluedevilwizard@038ec0bfa8394f4d9707d2661115c76b.service: Failed with result 'exit-code'.

Sep 07 16:23:20 fedora plasmashell[2555]: qrc:/qt/qml/plasma/applet/org/kde/plasma/systemtray/ExpandedRepresentation.qml:239: TypeError: Cannot read property 'text' of null
Sep 07 16:23:24 fedora kernel: input: soundcore Life Q30 (AVRCP) as /devices/virtual/input/input26
Sep 07 16:23:24 fedora systemd-logind[1477]: Watching system buttons on /dev/input/event19 (soundcore Life Q30 (AVRCP))
Sep 07 16:23:29 fedora plasmashell[2555]: QIODevice::read (QSslSocket): device not open
Sep 07 16:23:38 fedora chronyd[1443]: Selected source 23.143.196.204 (2.fedora.pool.ntp.org)
Sep 07 16:23:39 fedora kwin_wayland[2353]: Libinput: event5  - SynPS/2 Synaptics TouchPad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.31.3/touchpad-jumping-cursors.html for details
Sep 07 16:23:39 fedora kwin_wayland[2353]: Libinput: event5  - SynPS/2 Synaptics TouchPad: kernel bug: Touch jump detected and discarded.
See https://wayland.freedesktop.org/libinput/doc/1.31.3/touchpad-jumping-cursors.html for details
Sep 07 16:23:39 fedora systemd[2027]: app-org.kde.konsole-5904.scope: Consumed 6.385s CPU time over 3min 23.456s wall clock time, 30.7M memory peak.
Sep 07 16:23:40 fedora systemd[2027]: Started app-org.kde.konsole@86b2972e3db34ddf8e523d3db73c0194.service - Konsole - Terminal.
Sep 07 16:23:41 fedora systemd[2027]: Started app-org.kde.konsole-6287.scope.
Sep 07 16:23:46 fedora konsole[6287]: QObject::disconnect: Unexpected nullptr parameter
Sep 07 16:23:48 fedora systemd[2027]: Started app-org.kde.konsole@7980c6fef04b49298cce8ba1fc650372.service - Konsole - Terminal.
Sep 07 16:23:48 fedora systemd[2027]: Started app-org.kde.konsole-6405.scope.
Sep 07 16:23:55 fedora bluetoothd[1437]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Sep 07 16:23:55 fedora kded6[2520]: No object for name "bluez_input.98:47:44:A3:1A:DE"
Sep 07 16:23:55 fedora kdeconnectd[2860]: No object for name "bluez_input.98:47:44:A3:1A:DE"
Sep 07 16:23:55 fedora plasmashell[2555]: No object for name "bluez_input.98:47:44:A3:1A:DE"
Sep 07 16:23:55 fedora kded6[2520]: No object for name "bluez_output.98_47_44_A3_1A_DE.1"
Sep 07 16:23:55 fedora kdeconnectd[2860]: No object for name "bluez_output.98_47_44_A3_1A_DE.1"
Sep 07 16:23:55 fedora plasmashell[2555]: No object for name "bluez_output.98_47_44_A3_1A_DE.1"
```
Output from re-pairing/connecting with bluetoothctl:

```
<bluetoothctl]> pair 98:47:44:A3:1A:DE
Attempting to pair with 98:47:44:A3:1A:DE
[CHG] Device 98:47:44:A3:1A:DE Connected: yes
[CHG] Device 98:47:44:A3:1A:DE Bonded: yes
[CHG] Device 98:47:44:A3:1A:DE UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device 98:47:44:A3:1A:DE UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 98:47:44:A3:1A:DE UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 98:47:44:A3:1A:DE UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 98:47:44:A3:1A:DE UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device 98:47:44:A3:1A:DE UUIDs: 66666666-6666-6666-6666-666666666666
[CHG] Device 98:47:44:A3:1A:DE UUIDs: df21fe2c-2515-4fdb-8886-f12c4d67927c
[CHG] Device 98:47:44:A3:1A:DE ServicesResolved: yes
[CHG] Device 98:47:44:A3:1A:DE Paired: yes
Pairing successful
[DEL] Device 4F:B1:CF:D6:7D:49 4F-B1-CF-D6-7D-49
[DEL] Device CB:A7:19:12:81:91 LHB-8564DCCD
[DEL] Device B8:8C:29:CA:EC:55 net
[DEL] Device 57:E9:DD:EA:1C:89 57-E9-DD-EA-1C-89
[DEL] Device 5C:E7:51:64:DC:13 5C-E7-51-64-DC-13
[DEL] Device 76:00:61:F6:05:6C 76-00-61-F6-05-6C
[DEL] Device F8:0E:2E:E9:F6:DF LHB-B9C3F8FB
[... cut ...]
[SIGNAL] Disconnected - org.bluez.Reason.Unknown, Unspecified
[CHG] Device 98:47:44:A3:1A:DE Connected: no
[DEL] Transport /org/bluez/hci0/dev_98_47_44_A3_1A_DE/sep2/fd0
[DEL] Endpoint /org/bluez/hci0/dev_98_47_44_A3_1A_DE/sep1
[DEL] Endpoint /org/bluez/hci0/dev_98_47_44_A3_1A_DE/sep2
```

Since installing 5.80 didn't fix the issue for me, I tried a build of bluez with a reverted commit:

https://copr.fedorainfracloud.org/coprs/ethanej/Soundcore-Bluez/

but that didn't fix it. I tried removing KDE Connect, that didn't fix it.

ValdikSS, 2026-09-09:
Just a shot in the dark — have you tried cleaning `/var/lib/bluetooth/<your-card-bdaddr>/<your-headphones-braddr>`?

yaomtc, 2026-09-09:
I have not, I'll try that next.

Also, I found bluetooth-hcidump. Got a lead:

```
$ hcidump  
HCI sniffer - Bluetooth packet analyzer ver 2.5 
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff 
> HCI Event: Disconn Complete (0x05) plen 4 
    status 0x00 handle 11 reason 0x22 
    Reason: LMP Response Timeout
```

Not sure what LMP is, but I did find a thread with Arch Linux users claiming some firmware fixed a disconnection issue with a TP-Link adapter which, from what I can tell, uses a specific Realtek chip. I can't tell exactly what Realtek chip I have, though, all I have is its ID.

https://bbs.archlinux.org/viewtopic.php?id=302528

https://aur.archlinux.org/packages/rtl8761b-firmware

yaomtc, 2026-09-09:
@ValdikSS After having the thinkpad forget the headphones, its ID no longer appears in that directory. Thanks though!

ValdikSS, 2026-09-09:
For the statistics, I have Intel AX211 and use different bluetooth headphones with it, no issues. KDE, Fedora 43.
I was subscribed to this ticket due to somewhat similar issue which turned to be in pipewire and is long fixed.
So it indeed could be a firmware issue or a particular incompatibility in HCI stack of your exact chip and BlueZ.

Ne0W0r1d, 2026-10-20:
Yeah, AX211 is working for me now(Fc44 5.87+1.git8750129efca8) . But I don't know why playing while a time and will be laggy, need to resetting decode mode to sbc and back to aac.🤔

But nah, It will not drop the connection now,  Thx!